milaq/rpi-rf #1: Wiring diagram

tedstriker, 2016-04-21:
Hi Milaq,

could you please provide a wiring diagram, how to set things up?
Would make it a lot easier for us.
Thank you,
Ted

milaq, 2016-04-21:
Quick and dirty for a **Raspberry Pi 1/2 B+**:
```
                      RPI [B+]
              ____________
             |        ____|__
             |       |    |  |
             |     01|  . x  |02
             |       |  . x__|________       RX
             |       |  . x__|______  |   ________
             |       |  . .  |      | |  |        |
   TX        |   ____|__x .  |      | |__|VCC     |
 _______     |  |  __|__x .  |      |    |        |
|       |    |  | |  |  x____|______|____|DATA    |
|    GND|____|__| |  |  . .  |      |    |        |
|       |    |    |  |  . .  |      |    |DATA    |
|    VCC|____|    |  |  . .  |      |    |        |
|       |         |  |  . .  |      |____|GND     |
|   DATA|_________|  |  . .  |           |________|
|_______|            |  . .  |
                     |  . .  |
                     |  . .  |
                     |  . .  |
                     |  . .  |
                     |  . .  |
                     |  . .  |
                   39|  . .  |40
                     |_______|

TX:
   GND > PIN 09 (GND)
   VCC > PIN 02 (5V)
  DATA > PIN 11 (GPIO17)

RX:
   VCC > PIN 04 (5V)
  DATA > PIN 13 (GPIO27)
   GND > PIN 06 (GND)
```

Will add this (or maybe a prettier version?) to the readme, I'll keep the issue open until then.
Need to catch some sleep first though.

Hope it helps,
cheers

tedstriker, 2016-04-23:
thanks for your quick reply! This is totally fine.

milaq, 2016-04-23:
you're welcome.

resolved with 49097fb50561cf2b890e0ae1b9c95814c65057c2

closed 2016-04-23 by milaq