milaq/rpi-rf #2: Is this compatible with  Pi 3?

JoeFromQA, 2016-09-27:
Hi,

Having some issues using this with Home Assistant, just would like to rule out if it could be that I am using a Pi 3, opposed to the Pi 2 this was originally made for. Any help would be appreciated, thanks!

milaq, 2016-09-28:
The GPIO pinout is the same for the RPi 2 and 3. So the wiring shouldn't pose a problem.

Nevertheless, there _may_ be timing issues. I have no Raspberry Pi 3 to test this atm.
If you're having issues receiving data, you can try to adjust `rx_tolerance`.

Other than that, there should be no problems that come to my mind tight now.

Landrash, 2016-09-29:
@milaq I got access to a rpi3 and the necessary hardware . I've seen a few of these question come up in the Home Assistant chat and forums. 
What I've seen have been users issuing commands with protocol 1 and it being sent as protocol 5.
Workaround's I've seen have been to use the https://github.com/ninjablocks/433Utils instead.

JoeFromQA, 2016-09-30:
I believe my issue is transmitting out, and possibly the pulse length.
Here you can see the data I am getting when listening to the remote of the RF outlet, compared to listening to the Tx (thats also on the rpi) The codes are the same, but the pulse length is about twice as much. 
http://oi68.tinypic.com/2ytpkj9.jpg
I have tested this with 433utils, Rf_swtich and transmitting via Home Assistant with the same results. Also bought new transmitter thinking mine could have been bad, but I have yet to get the outlet to turn on/off from the rpi.   Thanks for the feedback, if I ever get it working I will post my results, maybe it'll help someone else.

milaq, 2016-10-04:
Thanks,
I'll have a look at it and try to reproduce it with 433utils/rc-switch on the transmitting end.

n8henrie, 2016-12-06:
This package works brilliantly on my Pi3. My switches are protocol 1, pulse length 190.

@JoeFromQA I've put together some of the 433utils tools and fairly thorough instructions at <https://github.com/n8henrie/rf_pi>, just going through my process in figuring out the basics of getting my 433 setup working on the Pi, a basic C++ example script (which I use through homeassistant as a `switch.command_line`), and a Python example using a shared library.

It may be an alternative to help you test / debug, but to be honest I found @milaq's repo a few weeks ago and much prefer it -- especially as it seems to be very reliable without employing the scheduling manipulation I've used in my scripts.

bfqrst, 2016-12-12:
Hey @milaq , when you say 

> adjust rx_tolerance

what exactly do you mean by that? Up or down, multiples of 80?

I have trouble getting any reading at all with my outlet / receiver combo. I got a pack of three cheap Aukru branded sender/receiver from Amazon. Same result with all of them. I can switch them up and down with the remote, but nothing in terms of sniffing. Changed GPIOs, tried a PI Zero and a PI 3, nada...

Could the reception be THAT bad without those 7cm wire antennas soldered to them?

Cheers
Ralph

n8henrie, 2016-12-16:
@bfqrst -- 

- What programs have you tried for sniffing? I use a similar module, and RFSniffer has always worked without hassle.
- What distance are you from the receiver when you click the remote? You can be both too far or also too close with some of these.
- How much time are you waiting after clicking? In one of these 433 repos I saw an issue raised by someone who just wasn't waiting enough time after starting the sniffer until clicking, or after clicking to display the info.

bfqrst, 2016-12-16:
Hey @n8henrie, thanks for getting back to me.

Well I tried this very rpi-rf which seemed most straightforward and easy enough to set up. I also compiled the ninjablocks Pi sniffer with the same result.
In terms of distance I clicked the remote literally next to that coil antenna as well as 1 meter to a couple of meters away. I tried really fast clicks and long presses. How much time would you advise to wait after one click? 
On a side note, tomorrow I expect a 5 pack of Etekcitys which are known to work. This way I'll be able to narrow it down to receiver vs outlets...

Ralph

n8henrie, 2016-12-16:
Perhaps you should move this to a new issue?

I'd recommend: 

1. Make sure only the receiver is plugged in to the Pi, leave the transmitter off / disconnected
1. Start `RFSniffer` from 433utils or from my rf_pi repo linked above. I recommend RFSniffer because it seems to be the "tried and true" method used by many for several years. rpi-rf is awesome and I'm thrilled to have a pure python solution, but it's relatively new on the scene. I recommend trying to get set up the old fashioned way for initial debugging.
1. Wait 10 seconds
1. Click and release button on your remote
1. Wait 5 seconds
1. Repeat 5 times

Regarding the readings you got: this one looks legit: `2016-12-16 16:41:20 - [INFO] 433: 524288 [pulselength 198, protocol 3]`. Could there be some nearby noise?

Hopefully the etekcity gives you better results -- that's what I'm using, no issues from day one.

closed 2018-08-10 by milaq