milaq/rpi-rf #3: Support for C.H.I.P.

chrisvis, 2016-10-01:
Hi,

First of all, great library! I use it to control my porch light through a component in [Home Assitant](https://home-assistant.io). I recently migrated my setup from a RPi to a [C.H.I.P.](https://getchip.com) and noticed that this package does not work anymore. This because `RPi.GPIO ` of course only works on a RPi. However there is a (almost) drop-in replacement for this dependency called [CHIP_IO](https://github.com/xtacocorex/CHIP_IO). The only method that CHIP_IO seems to lack is `GPIO.setmode`.

With this patch a got the library working on the C.H.I.P.:

```diff
diff --git a/rpi_rf/rpi_rf.py b/rpi_rf/rpi_rf.py
index 3949169..72c1e70 100644
--- a/rpi_rf/rpi_rf.py
+++ b/rpi_rf/rpi_rf.py
@@ -6,7 +6,10 @@ import logging
 import time
 from collections import namedtuple

-from RPi import GPIO
+try:
+    from RPi import GPIO
+except:
+    from CHIP_IO import GPIO

 MAX_CHANGES = 67

@@ -54,8 +57,11 @@ class RFDevice:
         self.rx_proto = None
         self.rx_bitlength = None
         self.rx_pulselength = None
-
-        GPIO.setmode(GPIO.BCM)
+        # the method 'setmode' is not available in CHIP_IO.GPIO
+        try:
+            GPIO.setmode(GPIO.BCM)
+        except:
+            pass
         _LOGGER.debug("Using GPIO " + str(gpio))

     def cleanup(self):
```

If you are willing to add support for C.H.I.P. I could make a PR out of this patch.

Cheers!

bachp, 2016-10-30:
I think it would make more sense to make the module injectable by some means. This could then also be used for Beaglebone black with [Adafruit_BBIO](https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/gpio).

chrisvis, 2016-11-02:
Yes, I agree that using injection would be a cleaner approach.
I can imagine that an important requirement is to stay backwards compatible, so that the default gpio module would be RPi version.

nunojusto, 2017-01-04:
Hello guys, i started porting rpi-rf (that i love) to CHIP using the new CHIP_IO.GPIO from @xtacocorex
It's working now... or sort of... you can check it out here https://github.com/nunojusto/CHIP-rf/

The problem is that i cannot have clear results and i', working directly with CHIP_IO guy to solve some of the error.
Please try and help if you can :)
Thankx

@bachp @chrisvis @milaq

xtacocorex, 2017-01-04:
The setmode() function isn't supported on the BBB either (CHIP_IO is ported from it), so the try/except catch you have @chrisvis is awesome.  Which GPIO pins were you using @chrisvis when you got this working?  Since this library uses callbacks, only the XIO and AP-EINT pins can be used on the CHIP.

If needed, I can always add the setmode() into CHIP_IO and have it do nothing.

nunojusto, 2017-01-04:
Also @chrisvis, i see that AP-EINT PINs are the only ones who can give a good response. XIO have much more interference. And also, 3,3v for receiver, works better than 5v.