milaq/rpi-rf #4: Lots of noise versus 433Utils version...

shendrick, 2016-12-31:
I'm trying to run the Raspberry Pi GPIO RF Switch component on home-assistant but when trying to use rpi-rf_recieve, I get lots of noise when pressing down the remote button...

<pre>
pi@raspberrypi:~ $ sudo rpi-rf_receive
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 505568 [pulselength 79, protocol 3]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 243428 [pulselength 80, protocol 3]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 131136 [pulselength 398, protocol 5]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 521967 [pulselength 81, protocol 3]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 505568 [pulselength 80, protocol 3]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 131072 [pulselength 411, protocol 5]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 15 [pulselength 80, protocol 3]
2016-12-31 16:50:06 - [INFO] rpi-rf_receive: 505568 [pulselength 79, protocol 3]
</pre>

Here's what I get with RFSniffer included with 433Utils...

<pre>
pi@raspberrypi:~/433Utils/RPi_utils $ sudo ./RFSniffer
Received 4543804
Received 4543804
Received 4543804
Received 4543804
Received 4543804
Received 4543804
Received 4543804
</pre>

Likewise when sending the code via rpi-rf_send nothing happens, but it works with 433Utils send command. 

Any help would be greatly appreciated!

falcomnl, 2017-02-03:
Same problem here, same symptoms. Using [this](http://www.ebay.nl/itm/433MHz-Superheterodyne-RF-Link-kits-3400-ARM-MCU-Transmitter-and-Reveiver-/281169560721?pt=LH_DefaultDomain_0&hash=item4177030491) Transmitter and Reveiver on a Raspberry Pi 2 w. Home Assistant

uh-nmb, 2017-02-09:
Same, I also noticed it was chewing up about 60% cpu on my pi

milaq, 2017-03-02:
There may be some timing issues with using Python instead of bare C code.
Unfortunately, GPIO flipping is quite CPU intensive (especially on the Pi 1).

If I have some spare time left, I'll have a look at possible time/cpu tradeoff optimizations.

flatsiedatsie, 2018-01-26:
Any update on this?

deadrop, 2018-06-22:
@milaq and about the filter (protocols) used by 433Utils? When I run ./RFSniffer the receive code is stable and continuos. How can I do the same through of rpi-rf_receive?

turbo-lab, 2018-12-26:
Try my solution: https://github.com/milaq/rpi-rf/pull/38

milaq, 2019-01-23:
Please confirm, if the latest master (specifically https://github.com/milaq/rpi-rf/commit/3bbd31cc3bbd4f1a7c2aba776d2541b37480e45b) improve the situation.

Thanks and cheers

flatsiedatsie, 2019-01-24:
I ended up building a universal 433Mhz signal recogniser/replayer.
https://forum.mysensors.org/topic/10094/candle-433-hub-a-universal-433mhz-signal-cloner/3

enesbcs, 2019-02-12:
I think that in some cases it is better to watch for only one protocol. At least it is a very nice solution to let the user decide if sniffing is the main goal, or watching for only protocol 1. (passed by a parameter to __init__)
It looks a bit smoother:
```
2019-02-12 22:35:58 - [INFO] rectest: Listening for codes on GPIO 17
2019-02-12 22:36:03 - [INFO] rectest: 4593709 [pulselength 551, protocol 1]
2019-02-12 22:36:05 - [INFO] rectest: 4593709 [pulselength 549, protocol 1]
2019-02-12 22:36:07 - [INFO] rectest: 4593709 [pulselength 548, protocol 1]
2019-02-12 22:36:12 - [INFO] rectest: 4593709 [pulselength 548, protocol 1]
```
https://github.com/enesbcs/rpi-rf/commit/6f6018d933c1daefe4e0e5404e860f274a85cb39