milaq/rpi-rf #6: Not receiving from any RF remote

Cloudore, 2017-02-19:
Hey everyone,
So my last project was to do what is described on this brush bruh automation video(albeit with rf shades instead of plugs):
https://www.youtube.com/watch?v=5UUazFbK-Hg&t
However, when I get to "sniffing" the codes from my rf remote, it doesn't really work. I press the button on my controller while running the rfrx.py/RFSniffer/rpi-rf_receive(Sniffs on GPIO 27) script and nothing happens, it does record some random codes, but they're not the ones being sent from my remote
When I deploy codes from my transmitter, however, it does seem to be receiving them on the rfrx.py/RFSniffer/rpi-rf_receive script, which leads me to believe it's working properly.
http://imgur.com/a/DnfgD
This is how I have the pi wired:
http://imgur.com/a/mTQHE
and one of the controllers I'm trying to emulate:
http://imgur.com/a/f87PL
I've already tried everything I could think so without much success, so if anyone has any guidance I'd appreciate it!
Also just realized that if I send through the transmitter a code 10 digits long, it won't show on the receiving side

sebastienvermeille, 2017-03-15:
@Cloudore did you solve it ? Same problem for me now ^^

Cloudore, 2017-03-15:
Yes, did so using this instead https://github.com/jderehag/rpi-rfsniffer.

Worked properly. Do mind that one is used differently, you record and play the RF codes. So I made a bash script that executed the command "rfsniffer play x" and then made a .php that said "execute bash script x", that allowed me to create a web interface for my rf controlled devices. Make sure apache has the proper permissions to run a bash script if you do go the way I did.

closed 2018-03-20 by milaq