milaq/rpi-rf #9: Add support for pigpio

matthijskooijman, 2017-05-08:
The pigpio library/daemon uses the DMA controller for very accurate hardware-based pin sampling and timestamping. This allows very exact reception (and probably also transmission, but I didn't look at that yet), unlike the GPIO-callbacks used now. Pigpio uses a daemon written in C that talks to the hardware, and exposes an interface using a TCP socket or FIFO/pipe, of which the former is used by a python wrapper they also provide.

I did a very quick implementation of RX, which seems to work well. Here's what I get with the unmodified library:
```
2017-05-08 16:12:22 - [INFO] rpi-rf_receive: 5326161 [pulselength 321, protocol 1]
2017-05-08 16:12:22 - [INFO] rpi-rf_receive: 5326161 [pulselength 320, protocol 1]
2017-05-08 16:12:22 - [INFO] rpi-rf_receive: 5326161 [pulselength 320, protocol 1]
2017-05-08 16:12:22 - [INFO] rpi-rf_receive: 5326161 [pulselength 319, protocol 1]
```

Here's what I get when using the unmodified library with `apt-get update` (or something similar with a moderate load) running in the background:

```
2017-05-08 16:16:42 - [INFO] rpi-rf_receive: 12288 [pulselength 706, protocol 5]
2017-05-08 16:16:42 - [INFO] rpi-rf_receive: 5326161 [pulselength 321, protocol 1]
2017-05-08 16:16:42 - [INFO] rpi-rf_receive: 8320 [pulselength 647, protocol 5]
2017-05-08 16:16:43 - [INFO] rpi-rf_receive: 5326161 [pulselength 323, protocol 1]
2017-05-08 16:16:43 - [INFO] rpi-rf_receive: 5326161 [pulselength 140, protocol 3]
2017-05-08 16:16:43 - [INFO] rpi-rf_receive: 5326161 [pulselength 327, protocol 1]
2017-05-08 16:16:43 - [INFO] rpi-rf_receive: 5326161 [pulselength 325, protocol 1]
2017-05-08 16:16:43 - [INFO] rpi-rf_receive: 8192 [pulselength 1652, protocol 4]
2017-05-08 16:16:43 - [INFO] rpi-rf_receive: 5326161 [pulselength 321, protocol 1]
```

When some big wget is running in the background (so lots of USB activity, I suspect), I get random output when not sending anything, and pretty much the same when sending something.

When using pigpio, the output is flawless, even with apt-get or wget running (though with wget, the output is delayed, probably because there's two different processes involved). I think this change would also solve #4.

I've attached my current patch below, to use it make sure to install pigpio first (there's a `python3-pigpio` Debian package) and run the daemon using `pigpiod -g -l` (-g: run in foreground, -l: local connections only).

This is just a prototype patch, to properly implement this the low-level I/O code should be separated from the decoding code so you can have multiple I/O backends (which makes this related to #3).

```
diff --git a/rpi_rf/rpi_rf.py b/rpi_rf/rpi_rf.py
index 3949169..bf65c43 100644
--- a/rpi_rf/rpi_rf.py
+++ b/rpi_rf/rpi_rf.py
@@ -6,7 +6,8 @@ import logging
 import time
 from collections import namedtuple

-from RPi import GPIO
+#from RPi import GPIO
+import pigpio

 MAX_CHANGES = 67

@@ -48,6 +49,7 @@ class RFDevice:
         self._rx_last_timestamp = 0
         self._rx_change_count = 0
         self._rx_repeat_count = 0
+        self._rx_handler = None
         # successful RX values
         self.rx_code = None
         self.rx_code_timestamp = None
@@ -55,7 +57,8 @@ class RFDevice:
         self.rx_bitlength = None
         self.rx_pulselength = None

-        GPIO.setmode(GPIO.BCM)
+        self.pigpio = pigpio.pi()
+
         _LOGGER.debug("Using GPIO " + str(gpio))

     def cleanup(self):
@@ -65,7 +68,6 @@ class RFDevice:
         if self.rx_enabled:
             self.disable_rx()
         _LOGGER.debug("Cleanup")
-        GPIO.cleanup()

     def enable_tx(self):
         """Enable TX, set up GPIO."""
@@ -164,24 +166,25 @@ class RFDevice:
             return False
         if not self.rx_enabled:
             self.rx_enabled = True
-            GPIO.setup(self.gpio, GPIO.IN)
-            GPIO.add_event_detect(self.gpio, GPIO.BOTH)
-            GPIO.add_event_callback(self.gpio, self.rx_callback)
+            self.pigpio.set_mode(self.gpio, pigpio.INPUT)
+            def callback(gpio, level, tick):
+                self.rx_callback(tick)
+            self._rx_handler = self.pigpio.callback(self.gpio, pigpio.EITHER_EDGE, callback)
             _LOGGER.debug("RX enabled")
         return True

     def disable_rx(self):
         """Disable RX, remove GPIO event detection."""
         if self.rx_enabled:
-            GPIO.remove_event_detect(self.gpio)
+            self._rx_handler.cancel();
+            self._rx_handler = None
             self.rx_enabled = False
             _LOGGER.debug("RX disabled")
         return True

     # pylint: disable=unused-argument
-    def rx_callback(self, gpio):
+    def rx_callback(self, timestamp):
         """RX callback for GPIO event detection. Handle basic signal detection."""
-        timestamp = int(time.perf_counter() * 1000000)
         duration = timestamp - self._rx_last_timestamp

         if duration > 5000:
```

flatsiedatsie, 2018-01-20:
This really made it more predictable. Thanks!

milaq, 2018-08-29:
may be relevant to #20

milaq, 2018-08-29:
Thanks @matthijskooijman for your input.
This issue slipped my attention for some reason.

I'll have a look at it ASAP

flatsiedatsie, 2018-10-25:
I tried it and it works great.

MattyBoy4444, 2018-11-15:
What is the status of this patch?

Esprim0, 2022-01-29:
> I implemented this change. It still uses over 30% cpu after the enable_rx() is called. Plus now you have the overhead of pigpiod, which is around 5-8%. Has anyone else had the same experience?

Same here! Maybe is there any different solution to get the 433 mhz work on a raspberry pi?