milaq/rpi-rf #10: Using rpi-rf with OrangePI

carlosfelipe29, 2017-07-18:
HI everybody,

Could I use this python package rpi-rf with an Orange Pi,  because Orange Pi works with raspbian, and other SO that are compatible with Raspberry.

Is there another Python Module to use in OrangePi which be able to read the rf433 Mhz.

I appreciate your help

rkarlsba, 2018-03-01:
I would like this very much too. I have a few opi zeros, and they look like a nice thing to use with home assistant, weren't it for the lack of things like rpi-rf

ec-blaster, 2018-05-18:
I wonder if you could substitute RPi-GPIO with [OPi.GPIO](http://opi-gpio.readthedocs.io/en/latest/index.html).
I'm trying it but I have "permission denied" errors...