milaq/rpi-rf #11: Doesn't work in RPi1

Chrisly3ear, 2017-07-20:
I have a Raspberry Pi 1 Model B.

I tried to send codes via `codesend` from https://github.com/ninjablocks/433Utils and it works.
If I try to do the exact same thing (same GPIO, protocol, pulselength, and code) it doesn't.

Can't tell why. Maybe RPi.GPIO is broken, but I don't think so.

Chris

palto42, 2017-10-05:
On my Raspberry Pi 2-B it works for the default pulse length (Comag switches), but not for short pulselength of 188 microseconds (ZAP/REV switches). The C-program `codesend` works for default and short pulselength. Maybe Python is too slow in the older Pi for short pulse length?
@Chrisly3ear what code and pulselength have you tried?
I also can't use the receive sniffer for the sockets with the short pulselength.

palto42, 2018-04-27:
Hi @AsharLohmar 
I think you mean that it's not clear which pin numbering is used:
`parser.add_argument('-g', dest='gpio', type=int, default=27,
help="GPIO pin (Default: 27)")`
The term "pin" can be misinterpreted as BOARD/physical numbering, although it's using the GPIO/BCM numbering (too many different names and formats, there's also the wiringPi numbering).
GPIO 27 = BCM 27 = Board 13 = WiringPi 2 (on RPi1 rev.1 the Board 13 was GPIO 21)

Agree that it would be useful to update the help text to make it more clear the BCM numbering is used.
`parser.add_argument('-g', dest='gpio', type=int, default=27,
help="GPIO BCM number (Default: 27)")`