milaq/rpi-rf #13: Max number of bits sent?

JoseFarias, 2017-09-13:
Hi,

The sniffer detects a 10-digit (essentially, 32 bit) number coming from my remote. I have tried sending this signal over the transmitter to no avail. I suspect this is due to the tx_code method defined for the RFDevice class.

Specifically, I suspect the problem lies within line 105, wich reads:

`rawcode = format(code, '#0{}b'.format(self.tx_length + 2))[2:]`

tx_length gets a value of 24 by default. Which I think means the binary code transmitted is limited to 24 bits (not enough to transmit my code).

Before I make any experimentation (partly due to the fact that I don't have acces to my Raspberry Pi at the  moment) I wanted to ask if this makes any sense and if changing the default value of tx_length so something greater will help my cause.

Thanks in advance!

JoseFarias, 2017-09-16:
-SOLVED!-
Solution:
I just had to create my own code (i.e. not use the send.py script) and instantiate an RFDevice object with the length parameter set to 32.

Turns out, it did work. If this is happening to you be weary of the following:
The code I was getting could have 'fit' within 30 bits. So it was tempting to just type 30 as the length. However, the receiver was expecting 32-bit code (which is standard, I suppose) so typing 30 would definitely NOT work even if the whole code can be fit into 30 bits.

closed 2017-09-16 by JoseFarias

staraxis, 2017-11-04:
Hi, I know this issue is closed but would you be able to provide your code or what lines you changed to get this to work with 32 bit codes? I still have an issue, even though I changed the tx_length to 32 I am seeing incorrect codes being sent along with incorrect protocol and pulse length. I am running this through the RF Switch component on home assistant using a raspberry pi.

JoseFarias, 2017-11-04:
Certainly, @staraxis 

I didn't actually change any of the rpi-rf source code; since that is generally not recommended. What I did to make this work was to essentially ditch the send.py script included in the repository and made my own script that would send 32 bit codes (or any-size codes for that matter).

Here is an example of how you might go about doing this:

```
from rpi_rf import RFDevice
from time import sleep

class deviceClass(RFDevice):
    def __init__(self, gpio=17, protocol=1, pulse=300, repeat=10, length=1, tolerance=80, onCode=1, offCode=2):
        self.protocol = protocol
        self.pulse = pulse
        self.repeat = repeat
        self.length = length
        self.tolerance = tolerance
        self.onCode = onCode
        self.offCode = offCode
        RFDevice.__init__(self, gpio, protocol, pulse, repeat, length, tolerance)

    def turn_on(self):
        self.enable_tx()
        self.tx_code(self.onCode, self.protocol, self.pulse)

    def turn_off(self):
        self.enable_tx()
        self.tx_code(self.offCode, self.protocol, self.pulse)

myDevice = deviceClass(onCode = 1104643512, offCode = 1089042524, length = 32)

myDevice.turn_on()
sleep(5)
myDevice.turn_off()
myDevice.cleanup()

```
Essentially, we just created a deviceClass which inherits from the RFDevice class included in rpi-rf.

From my example, you can see I gave all parameters default values which can be overwritten when a device object is instantiated. This way you are able to virtually send codes of any size without modifying the source code.

Hope this helps!