milaq/rpi-rf #20: Reduce CPU usage

Naozumi2k, 2018-02-18:
Hi there!
Is there way to reduce CPU usage on rpi zero since receive script use about 50%-75% cpu power?
I want to use rf receiver script on a battery powered project so cpu usage is important for me.
Playing with sleep time in the main loop does not helps a lot for me :(
[image]

milaq, 2018-03-20:
The older Raspberries and the Zero really don't have that much CPU power to spare; and GPIO is very taxing for it (esp. with such short intervals as 433Mhz codes).
I'll leave this open for further review but at the moment I don't think this can be improved upon.

jwillmer, 2018-05-18:
This is a problem. I get down to 5-6% CPU usage on my rpi zero with [pigpio](https://github.com/joan2937/pigpio). I was looking into `rpi-rf` since I think even 5-6% is too much. 

For comparison: `pigpio` is watching **all** GPIO pins and needs 5-6% CPU usage to do so and with `rpi-rf` I can setup to only watch one GPIO pin.

milaq, 2018-08-29:
Remember that utilizing GPIO can be very CPU intensive (esp. on Raspberry platforms).
We already use an interrupt driven logic, but maybe there is something we can do to further optimize the CPU usage.
I'll leave this open until I have the time to take a look at it or someone has a good suggestion.

mohanisch, 2020-01-12:
Are there any updates regarding this issue?