milaq/rpi-rf #24: Long range advice

sekisek, 2018-04-05:
everything is working great for me 🙂
But I wanted to achieve longer range and bought …..

https://www.ebay.com/itm/CC1101-wireless-module-CC1100-upgraded-version-of-the-high-precision-external/191674193090?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

https://www.ebay.com/itm/XL1101-D01-Wireless-RF-Transceiver-Module-CC1101-CC1100-SI4432-433MHz/192369991980?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

https://www.ebay.com/itm/SX1278-433MHz-Wireless-Transceiver-Module-LoRa-SPSP-Long-Distance/332012370467?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

I have been trying to make them work with no sucuss so I started thinking maybe my approach is wrong.
Will that work? any advice?

thanks

frankymacster, 2018-05-29:
I tried listening on different GPIOs.
Could it be that this library doesn't support devices using [SPI](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus)?

milaq, 2018-08-29:
We are using GPIO edge detection to listen for RF input.
As you correctly pointed out, we do not have support for SPI atm.

_But if you want to utilize your CC1101 I can very much recommend [culfw](http://culfw.de/culfw.html).
You can build one on your own easily with a CC1101 and a ATmega328p. Just google for 'CULstick diy' or similar_

closed 2018-08-29 by milaq