milaq/rpi-rf #33: send error

hacktorious, 2018-08-23:
python3.6 rpi-rf_send 123456
/usr/local/lib/python3.6/site-packages/rpi_rf/rpi_rf.py:77: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.gpio, GPIO.OUT)
2018-08-23 09:53:36 - [INFO] rpi-rf_send: 123456 [protocol: default, pulselength: default, length: default, repeat: 10]
Traceback (most recent call last):
  File "rpi-rf_send", line 49, in <module>
    rfdevice.tx_code(args.code, args.protocol, args.pulselength, args.length)
TypeError: tx_code() takes from 2 to 4 positional arguments but 5 were given

rebhuhnweg, 2018-08-25:
I've got the same Problem here.

hacktorious, 2018-08-28:
Just roll back to 0.9.5.  That fixed it for me.

rebhuhnweg, 2018-08-28:
also tried to go back. Uninstalled rpi-rf by pip3 and after that installed with

pip3 install rpi-rf==0.9.5

He installed rpi-rf 0.9.5 and rpi.gpio 0.6.3

Same problem :(

rebhuhnweg, 2018-08-28:
Fixed it :)

remove the last parameter (length) in this line in the send.py

rfdevice.tx_code(args.code, args.protocol, args.pulselength, args.lenght)

hacktorious, 2018-08-28:
Yes, sorry I should have mentioned that.

milaq, 2018-08-29:
duplicate of #31

closed 2018-08-29 by milaq