milaq/rpi-rf #48: send.py disables other remote controls

Al-, 2020-02-29:
Hi

I encountered an issue with the python send script, described below. My workaround works for me. Nevertheless, you may consider to modify the script.

Cheers

Al_

**Problem statement**:
After successfully installing and testing a cheap, no-name 433 MHz sender (attached to RPi 3, raspbian / stretch) to switch a relais, several completely independent hand-held remote controls (for the garage door and for blinds) stopped working.

**Identifying the issue**:
- After a fresh RPi reboot, the hand-held remote controls function properly.
- Once I have sent a command to the relais using the python send script, the hand-held remote controls permanently no longer function (failure confirmed over more than 24 hours)
- If I unplug the signal pin (in my case gpio 27), the hand-held remote controls again function correctly. Same if I unplug the power from the 433 MHz sender.

I expect that the 433 MHz sender continuously sends noise interfering with other remote controls using the same frequency. I further expect that this happens because the python send script leaves gpio 27 on '1' or undefined (in which case it may fluctuate between '0' and '1' in the absence of a pull-down resistor)

**Workaround**:
I wrap the call to the python send script in the bash script shown below. This solves the issue for me.
```
#!/bin/bash

#  -g GPIO         GPIO pin (Default: 17)
#  -p PULSELENGTH  Pulselength (Default: 350)
#  -t PROTOCOL     Protocol (Default: 1)

declare -r -i gpio=27

echo "executing as $( whoami )"

[ -d "/sys/class/gpio/gpio${gpio}" ] && echo ${gpio} > /sys/class/gpio/unexport && echo "unexported ${gpio} to /sys/class/gpio/"

~/send_433MHz.py -g ${gpio} -p 318 -t 1 $1

# it seems that the GPIO pin and/or the 433 MHz sender is not properly closed
# --> force GPIO pin to 0 (hoping that it is the pin, not the sender itself)
[ -d "/sys/class/gpio/gpio${gpio}" ] || echo ${gpio} > /sys/class/gpio/export && echo "exported ${gpio} to /sys/class/gpio/"
sleep 3s   # something, possibly udev rule, takes time to set the owner:group on /sys/class/gpio/gpio${gpio}/*
ls -lA /sys/class/gpio/gpio${gpio}/direction
echo "out" > /sys/class/gpio/gpio${gpio}/direction
ls -lA /sys/class/gpio/gpio${gpio}/value
echo 0 > /sys/class/gpio/gpio${gpio}/value
```
**Suggestion**
To modify the python script and force the signal gpio pin to '0' before ending the script.

intelwolf, 2020-03-05:
I have the same issue indeed, after using the script to transmit codes it disables other remote controls. I removed the the cleanup() routine and added GPIO.setwarnings(False) in the script  that fixed the issue also

paf0186, 2022-01-07:
The issue here is the cleanup() routine in the library - It needs to be called on a specific pin rather than without argument.