milaq/rpi-rf #60: Rpi-rf not working on Pi Zero W

juergen2345, 2021-12-24:
Hi,
i changed my pi4 to a zero w, and rpi-rf does not work.
I use Rasbian Buster on both Pi‘s.
Are there any known issues with a zero w?
Thx

paf0186, 2022-01-07:
It would help to say more about what "does not work" means.  What are you trying to do and what happens?

juergen2345, 2022-01-07:
Sorry.
I tried to use it on a Pi Zero to switch Power Sockets 

https://www.amazon.de/Brennenstuhl-Funkschalt-Set-Funksteckdosen-Funksteckdose-Kindersicherung/dp/B07CGBRS7T/ref=sr_1_10?__mk_de_DE=ÅMÅŽÕÑ&crid=Y24XGPR442ZN&keywords=brennenstuhl+steckdosen+funk&qid=1641540816&sprefix=brennstuhl+steckdose+funk%2Caps%2C102&sr=8-10

I used:
    Pulslength 325
    Protokol 1
    Codelength 24
    repeat 10 and i also tried 100

nothing happend. 
On a pi3 or a pi4 everything works fine.
A year ago, i used a Zero with this setup for an other receiver (with Pulslength 650) and it worked too.
I think the Zero is not powerfull enough? 
Thx