milaq/rpi-rf #73: Any chance to work with gpiozero?

tkhkbn, 2023-12-05:
the script gives error in raspberry pi5

JoelKle, 2024-01-30:
I've forked the repo and replaced the RPi.GPIO with gpiozero. Now it works on my Pi 5.
New release 0.10.0

@milaq Maybe this is helpful?

tkhkbn, 2024-02-01:
> I've forked the repo and replaced the RPi.GPIO with gpiozero. Now it works on my Pi 5. https://github.com/JoelKle/rpi-rf/releases/tag/0.10.0 New release 0.10.0
> 
> @milaq Maybe this is helpful?

thank you!!!

hugolopes3d, 2024-02-03:
@JoelKle the forked version still doesn't work on Pi5 with the sample rpi-rf-receive.py file:

The output is this one
`Traceback (most recent call last):
  File "/home/xxxx/Scripts/recieve.py", line 28, in <module>
    rfdevice.enable_rx()
  File "/home/xxxx/.local/lib/python3.11/site-packages/rpi_rf/rpi_rf.py", line 186, in enable_rx
    self.rx_device.when_changed = self.rx_callback
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 123, in __setattr__
    raise AttributeError(
AttributeError: 'DigitalOutputDevice' object has no attribute 'when_changed'`

These are the libraries installed:

gpiozero                           2.0
rpi-rf                             0.10.0

Would you have any thoughts on this?

JoelKle, 2024-02-05:
Oh yeah I should not copy/paste code generated by ChatGPT without testing it properly...
I've commited once again to my repo. It doens't throw an error but in my tests it looks like there's a wrong RX code returned. Maybe you can test it as well.

I'm not an expert for this 433Mhz stuff...

bc-bjoern, 2024-03-27:
I've been struggling with the same issue for 2 days. @JoelKle you made my day! 
The latest version in your fork works like a charm. I tested the receiver only. Thanks!

bc-bjoern, 2024-03-29:
Well. I can see the same issue as in your tests. 

Pressing the same button all the time here:
`python foooooo
on
2024-03-29 14:03:09 - [INFO] foooooo: 1104 [pulselength 1225, protocol 2]
off
2024-03-29 14:03:24 - [INFO] foooooo: 4095 [pulselength 1225, protocol 6]
on
2024-03-29 14:03:26 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]
on
2024-03-29 14:03:27 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]
2024-03-29 14:03:29 - [INFO] foooooo: 31 [pulselength 724, protocol 6]
on
2024-03-29 14:03:43 - [INFO] foooooo: 1104 [pulselength 1230, protocol 2]
off
2024-03-29 14:03:44 - [INFO] foooooo: 4095 [pulselength 1230, protocol 6]
off
2024-03-29 14:03:45 - [INFO] foooooo: 4095 [pulselength 1224, protocol 6]
on
2024-03-29 14:03:46 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]
on
2024-03-29 14:03:47 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]
on
2024-03-29 14:03:47 - [INFO] foooooo: 1104 [pulselength 1223, protocol 2]
on
2024-03-29 14:03:48 - [INFO] foooooo: 1104 [pulselength 1224, protocol 2]
on
2024-03-29 14:03:49 - [INFO] foooooo: 1104 [pulselength 1225, protocol 2]
on
2024-03-29 14:03:50 - [INFO] foooooo: 1104 [pulselength 1224, protocol 2]
on
2024-03-29 14:03:51 - [INFO] foooooo: 1104 [pulselength 1227, protocol 2]
on
2024-03-29 14:03:51 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]
2024-03-29 14:03:54 - [INFO] foooooo: 3152 [pulselength 1226, protocol 2]
2024-03-29 14:03:55 - [INFO] foooooo: 7 [pulselength 732, protocol 6]
on
2024-03-29 14:03:55 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]
off
2024-03-29 14:03:57 - [INFO] foooooo: 1106 [pulselength 1228, protocol 2]
on
2024-03-29 14:03:58 - [INFO] foooooo: 1104 [pulselength 1224, protocol 2]
on
2024-03-29 14:04:03 - [INFO] foooooo: 1104 [pulselength 1231, protocol 2]
on
2024-03-29 14:04:04 - [INFO] foooooo: 1104 [pulselength 1226, protocol 2]`

Wehrum, 2024-05-18:
Hi to anyone reading this. I ran into the same problem trying to get this working. After installing the packages in the actual repository (not any forks) I ran into the `RuntimeError: Failed to add edge detection`.

I then went into the root user (sudo -i) and was able to run it with no issues. Not saying it will 100% fix the problem but worth a shot if you haven't.